mit-aera/FlightGoggles #105: A few questions I also asked on the forums...

SethKitchen, 2019-03-13:
1) Is there an event for when the drone finishes the course? How do you pull the number of gates passed (points) and time from it?

Should I be keeping track of every time it passes a gate and just finish when it reaches the last gate... that seems tedious.

2) Do we have to run with graphics or can we run silent?
 
3) Also how are you supposed to send commands -- I am currently just calling key.press('a') from python - I know there is a joy stick alternative, but can't I just send an event?

eatal, 2019-03-13:
Hi SethKitchen,

1. This was raised previously as #87. As mentioned there, the reporter is the only node that has access to these events, but you could make them available by adding a publisher right after Line 147 in reporter.py.

2. You can run without the FlightGoggles renderer by removing it from the launch file. You could also set the parameter `use_external_renderer` to prevent the renderer from starting, e.g. 
```
roslaunch flightgoggles teleopExample.launch use_external_renderer:=1
```
However, note that the renderer also simulates the IR marker and downward rangefinder measurements, so these will become unavailable if the renderer is not used.

3. Yes, you can directly send angular rate and thrust commands by publishing to the `/uav/input/rateThrust` topic. See also on the [wiki](https://github.com/mit-fast/FlightGoggles/wiki/ros-topics). For more information on how topics work and how to publish to them, you can also have a look here: [wiki.ros.org](https://wiki.ros.org/ROS/Tutorials/UnderstandingTopics).

If you are using your own nodes to publish the commands, it is often best to disable the manual control nodes by running your own launch script instead of `teleopExample.launch`. You can still use `core.launch` to launch the core FlightGoggles nodes.

Hope this helps

closed 2019-03-13 by eatal

SethKitchen, 2019-03-13:
Thanks this is super helpful -- seems kinda jank IR and downward rangefinder are linked to renderer and there is not a publisher already built in for goals -- 

I will add and send a pull request

eatal, 2019-03-13:
You're welcome! 

IR and rangefinder are dependent on the renderer, because these measurements are simulated using ray tracing in Unity. This way occlusions by other objects in the environment can be taken into consideration.