mit-aera/FlightGoggles #106: Keeps changing my video options

cmayer777, 2019-03-13:
Ubuntu 16.04 RTX 2070
FG randomly resets my displays causing me to need to reboot.
I have dual displays set both to 1920x1080. Sometimes the 2nd display disappears, and sometimes the 1st monitor is set to 3840x2160.  I lose the edges and cannot control my computer. The display settings is locked out from setting it back, even after FG is quit.

Winter-Guerra, 2019-03-13:
Hi Chris Mayer,

This sounds like it could be a GPU driver crash. However, more information is needed. Could you please post your player.log pre-crash? 

Best,
-Winter

cmayer777, 2019-03-14:
Nothing crashes.  It sets it when it runs.

varunmurali1, 2019-03-14:
Hi Chris,

It is hard to understand the cause of your issue without further information. Could you post your player.log regardless? Also, what version of the nvidia driver are you using? Are you using stereo or a monocular camera setup?

Winter-Guerra, 2019-03-15:
Closing due to lack of information.

closed 2019-03-15 by Winter-Guerra