mit-aera/FlightGoggles #107: clockscale yield different results

JamesLuoau, 2019-03-13:
Hi, I have a ROS rate object configured as 8 Hz, a test code that:

```
publish to /uav/input/rateThrust x,y,z,thrust
rate.sleep()
publish to /uav/input/rateThrust x,y,z,thrust
rate.sleep()
publish to /uav/input/rateThrust x,y,z,thrust
rate.sleep()
publish to /uav/input/rateThrust x,y,z,thrust
rate.sleep()
publish to /uav/input/rateThrust x,y,z,thrust
rate.sleep()
publish to /uav/input/rateThrust x,y,z,thrust
rate.sleep()
```

when I config `clockscale: 0.1 `it flys to the point I want, however if I change` clockscale: 1`, the result completely different. I have run it many times and have same result.
Also I can confirm that Rate object works well

Is this expected result?
Thanks

Winter-Guerra, 2019-03-14:
Hi @JamesLuoau,

I'm sorry that you're running into this problem. Could you please provide us with the following information to help with our diagnoses? 

* Are you trying to enable real-time simulation? If so, we recommend commenting out `/uav/flightgoggles_uav_dynamics/clockscale` and set `/use_sim_time` in your launchfile to `false`.
* For `/use_sim_time` = `true` and `/uav/flightgoggles_uav_dynamics/clockscale` = 0.1, we expect that `rostopic hz --wall-time /clock` should output `~96Hz`. Is this the case?
* For `/use_sim_time` = `true` and `/uav/flightgoggles_uav_dynamics/clockscale` = 1.0, we expect that `rostopic hz --wall-time /clock` should output `~960Hz`. Is this the case?
* For `/use_sim_time` = `false` we expect that `rostopic hz --wall-time /clock` should not output any messages. Is this the case?
* Are you publishing fixed rateThrust commands or performing closed loop control based on the output of the IR/range finder/image sensors?

Best, 
-Winter

Winter-Guerra, 2019-03-14:
Hi @JamesLuoau,

We just tested all of the scenarios listed above and can confirm that the clockscale setting is working as intended, including for clockscales faster than real time. However, we did notice something that might cause users some slight confusion. At faster than real time clockscales, the renderer requires a few seconds of time on startup to "catch up".

Best,
-Winter

JamesLuoau, 2019-03-14:
Hi, in my case, `<param name="/use_sim_time" value="true" />`

Here I recorded 3 videos which have
All videos shows rviz, the hz--wall-time for /clock, roll_pitch_yaw_thrust sent to topic `/uav/input/rateThrust`, the current position. you will see that exactly same command, when in clockscale 1.0, it end to 7.6, 16.5, 1.8 (x, y, z), where in `clockscale: 0.1`, it end up to 4.95, 29.19, 4.92. 
For everyone, I config the init parameter and killed `/uav/flightgoggles_uav_dynamics` which I asked before https://github.com/mit-fast/FlightGoggles/issues/84, maybe I'm doing that wrong? 

1. `clockscale: 1.0`
[video](https://youtu.be/ZrP5UdHxaAw)
2. `clockscale: 0.1`
[video](https://youtu.be/rve3Fqge93M)
3. `#clockscale: 1.0` clockscale
[video](https://youtu.be/6L4Se8Wy8rs)

Hope the videos helps

Winter-Guerra, 2019-03-15:
Hi @JamesLuoau,

It looks like you're also running the teleop nodes. These nodes are not meant to be run while autonomous scripts are also running. These teleop nodes could be trying to override your autonomous controls. Could you try again, but without the teleop nodes running?

Best,
-Winter

Winter-Guerra, 2019-03-16:
Hey @JamesLuoau, any updates to this issue?

SimpleUAVs, 2019-03-16:
I am having the same problem. Our PID loops need to retuned depending on the sim speed.

SimpleUAVs, 2019-03-16:
The problem seems to be /uav/sensors/downward_laser_rangefinder which is updated at the FPS rate which changes depending on the speed to SIM is run at inverted. IE, on my system (Dell t3600 1080i GPU) 60hz in SIM time (slow) and 23hz in real time.

JamesLuoau, 2019-03-17:
Hi, @Winter-Guerra 

I have downloaded newest code, clean & build. Simple run `roslaunch flightgoggles core.launch`. 

edit drone.ymal with commented out, clockscale: 1.0 and clockscale: 0.1, run my test code 3 times each.

```
Commented Out

2019-03-16 23:31:51,304 - INFO - 1970-01-01 00:00:00 position 2.32, 33.80, 2.55 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
2019-03-16 23:31:51,416 - INFO - 1970-01-01 00:00:00 position 2.32, 33.80, 2.60 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
2019-03-16 23:31:51,553 - INFO - 1970-01-01 00:00:00 position 2.32, 33.80, 2.67 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
2019-03-16 23:31:51,811 - INFO - 1970-01-01 00:00:00 position 2.32, 33.80, 2.77 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
2019-03-16 23:31:52,339 - INFO - 1970-01-01 00:00:00 position 2.33, 33.79, 2.90 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
2019-03-16 23:31:52,879 - INFO - 1970-01-01 00:00:01 position 2.33, 33.79, 3.06 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
2019-03-16 23:31:53,399 - INFO - 1970-01-01 00:00:01 position 2.33, 33.79, 3.25 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
2019-03-16 23:31:53,956 - INFO - 1970-01-01 00:00:01 position 2.33, 33.78, 3.46 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
2019-03-16 23:31:54,474 - INFO - 1970-01-01 00:00:01 position 2.33, 33.77, 3.70 roll_pitch_yaw_thrust 0.00, 3.00, 0.00, 11.81
2019-03-16 23:31:55,055 - INFO - 1970-01-01 00:00:01 position 2.33, 33.74, 3.96 roll_pitch_yaw_thrust 0.00, 3.00, 0.00, 11.81
2019-03-16 23:31:55,646 - INFO - 1970-01-01 00:00:01 position 2.33, 33.65, 4.23 roll_pitch_yaw_thrust 0.00, 3.00, 0.00, 11.81
2019-03-16 23:31:56,268 - INFO - 1970-01-01 00:00:01 position 2.33, 33.44, 4.48 roll_pitch_yaw_thrust 0.00, 3.00, 0.00, 11.81
2019-03-16 23:31:56,905 - INFO - 1970-01-01 00:00:01 position 2.33, 33.07, 4.65 roll_pitch_yaw_thrust 0.00, -3.00, 0.00, 11.81
2019-03-16 23:31:57,547 - INFO - 1970-01-01 00:00:02 position 2.33, 32.55, 4.72 roll_pitch_yaw_thrust 0.00, -3.00, 0.00, 11.81
2019-03-16 23:31:58,192 - INFO - 1970-01-01 00:00:02 position 2.33, 31.90, 4.72 roll_pitch_yaw_thrust 0.00, -3.00, 0.00, 11.81
2019-03-16 23:31:58,846 - INFO - 1970-01-01 00:00:02 position 2.33, 31.17, 4.68 roll_pitch_yaw_thrust 0.00, -3.00, 0.00, 11.81
2019-03-16 23:31:59,493 - INFO - 1970-01-01 00:00:02 position 2.34, 30.41, 4.66 roll_pitch_yaw_thrust -1.00, 1.00, 0.00, 11.81
2019-03-16 23:32:00,127 - INFO - 1970-01-01 00:00:02 position 2.35, 29.67, 4.67 roll_pitch_yaw_thrust -1.00, 1.00, 0.00, 11.81
2019-03-16 23:32:00,779 - INFO - 1970-01-01 00:00:02 position 2.38, 28.95, 4.70 roll_pitch_yaw_thrust -1.00, 1.00, 0.00, 11.81
2019-03-16 23:32:01,431 - INFO - 1970-01-01 00:00:02 position 2.45, 28.24, 4.75 roll_pitch_yaw_thrust -1.00, 1.00, 0.00, 11.81
2019-03-16 23:32:02,085 - INFO - 1970-01-01 00:00:02 position 2.58, 27.53, 4.80 roll_pitch_yaw_thrust -1.00, 1.00, 0.00, 11.81
2019-03-16 23:32:02,734 - INFO - 1970-01-01 00:00:03 position 2.79, 26.78, 4.84 roll_pitch_yaw_thrust -1.00, 1.00, 0.00, 11.81
2019-03-16 23:32:03,387 - INFO - 1970-01-01 00:00:03 position 3.07, 26.00, 4.84 roll_pitch_yaw_thrust -1.00, 1.00, 0.00, 11.81
2019-03-16 23:32:04,032 - INFO - 1970-01-01 00:00:03 position 3.45, 25.19, 4.78 roll_pitch_yaw_thrust -1.00, 1.00, 0.00, 11.81
2019-03-16 23:32:04,675 - INFO - 1970-01-01 00:00:03 position 3.90, 24.34, 4.64 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 7.81
2019-03-16 23:32:05,324 - INFO - 1970-01-01 00:00:03 position 4.39, 23.50, 4.38 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 7.81
2019-03-16 23:32:05,978 - INFO - 1970-01-01 00:00:03 position 4.90, 22.66, 4.04 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 7.81
2019-03-16 23:32:06,621 - INFO - 1970-01-01 00:00:03 position 5.43, 21.80, 3.62 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 7.81
2019-03-16 23:32:07,272 - INFO - 1970-01-01 00:00:03 position 5.96, 20.95, 3.14 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 7.81
2019-03-16 23:32:07,923 - INFO - 1970-01-01 00:00:04 position 6.47, 20.10, 2.62 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 7.81
2019-03-16 23:32:08,572 - INFO - 1970-01-01 00:00:04 position 6.96, 19.24, 2.07 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 7.81
2019-03-16 23:32:09,230 - INFO - 1970-01-01 00:00:04 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 7.81
2019-03-16 23:32:09,888 - INFO - 1970-01-01 00:00:04 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:32:10,532 - INFO - 1970-01-01 00:00:04 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:32:11,180 - INFO - 1970-01-01 00:00:04 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:32:11,831 - INFO - 1970-01-01 00:00:04 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:32:12,475 - INFO - 1970-01-01 00:00:04 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:32:13,127 - INFO - 1970-01-01 00:00:05 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:32:13,779 - INFO - 1970-01-01 00:00:05 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:32:14,423 - INFO - 1970-01-01 00:00:05 position 7.41, 18.38, 1.50 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81

2019-03-16 23:32:35,314 - INFO - 1970-01-01 00:00:00 position 2.32, 33.80, 2.55 roll_pitch_yaw_thrust 0.00, 0.00, 0.00, 11.81
[... cut ...]
2019-03-16 23:38:30,009 - INFO - 1970-01-01 00:00:04 position 7.43, 18.20, 1.30 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:38:31,265 - INFO - 1970-01-01 00:00:05 position 7.43, 18.20, 1.30 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:38:32,516 - INFO - 1970-01-01 00:00:05 position 7.43, 18.20, 1.30 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
2019-03-16 23:38:33,756 - INFO - 1970-01-01 00:00:05 position 7.43, 18.20, 1.30 roll_pitch_yaw_thrust 1.00, 0.00, 0.00, 10.81
```

eatal, 2019-03-18:
Hi @JamesLuoau,

To recreate your simulation scenario, I made a Python ROS node that uses a ROS rate object configured at 8 Hz and publishes the same control inputs as given by `roll_pitch_yaw_thrust` in your last post. Using the attached Python script I was able to verify that the vehicle trajectory is as expected, and independent of clockscale value.

[testcontroller.py.zip](https://github.com/mit-fast/FlightGoggles/files/2978486/testcontroller.py.zip)

eatal, 2019-03-18:
Hi @SimpleUAVs,

The publish rate of rangefinder measurements is indeed identical to the camera render rate. This is because rangefinder measurements are simulated in Unity using ray tracing. Renderer maintains 60 Hz with automatic clock scaling. To provide the user with options to best suit their hardware, we also provide manual clockscale.

When manually setting the clockscale value, the renderer tries to maintain 60 fps in sim time. However, (depending on available hardware) the achieved frame rate may be lower. When lowering the clockscale value the ratio sim time/wall time is lowered, so that a higher frame rate (up to 60 fps) can be achieved on a given system.

The clockscale adjustment is done automatically when the clockscale parameter is commented out: It is continuously adjusted such that rendering is achieved at 60 fps in sim time.

JamesLuoau, 2019-03-19:
Thanks @eatal , I have tested and attached my file. with times scale:1.0, the max difference is 2.59 in y, I assuming this is expected?

 My Code didn't set up the header, so I tested it also, file I used to print out current positon.
[testcontroller.py.zip](https://github.com/mit-fast/FlightGoggles/files/2982011/testcontroller.py.zip)

### clockscale Commented Out
"""
Run 1
position(15.71, -13.47, 6.49)

Run 2
position(15.52, -13.46, 6.63)

Run 3
position(15.38, -14.52, 6.91)

Run 1, without timestamp header
position(16.38, -17.80, 6.72)

Run2, without timestamp header
position(16.48, -17.99, 6.53)

Run3, without timestamp header
position(16.43, -18.02, 6.79)
"""

### clockscale: 1.0
"""
Run 1
position(15.97, -16.59, 6.82)

Run 2
position(16.09, -14.57, 6.71)

Run 3
position(15.34, -14.00, 6.77)

Run 1, without timestamp header
position(15.75, -12.93, 6.95)

Run 2, without timestamp header
position(16.79, -11.85, 5.41)

Run 3, without timestamp header
position(14.96, -12.39, 6.72)
"""

eatal, 2019-03-19:
Hi @JamesLuoau,

I ran the Python script that you attached to your last post multiple times with clockscale 0.1, 1.0, 3.0, and commented out. In every case, the trajectory is as expected and the print outs from the Python script are as expected (no more than a few cm distance between runs).

My test procedure is using the master branch without any adaptions. For each run, I first start the Python script (to make sure it is running when sim time reaches 5 sec.), next I run the following command to start Flightgoggles `roslaunch flightgoggles core.launch ignore_collisions:=true`. I used the script you posted without any changes, except that I commented out line 44 which is unnecessary since I relaunch Flightgoggles for each run anyway.

If manually killing and restarting the dynamics node, it is required that this is done such that no control inputs from the Python script are missed. To guarantee consistency between runs, I would recommend the test procedure outlined above.

closed 2019-03-19 by eatal

JamesLuoau, 2019-03-20:
Thanks @eatal 
I git pull / clean / build and ignore_collisions:=true, in p3 instances I can see the result are expected. Did the same thing in my p2 instances, same issue. 

For all other people, if you using p2, please have a test.

eatal, 2019-03-20:
Good to hear it's working on p3, @JamesLuoau.

Just to check: You are running both the uav dynamics and the Python script that generates the control commands on the AWS instance, correct? If that's not the case, network latency and dropped messages may be contributing to the errors.

For debugging, I would recommend recording a [rosbag](https://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data). This way it's straightforward to examine (e.g. using Python or MATLAB) exactly what commands are published to `/uav/input/rateThrust` and when (in sim time), and make sure they are the same between runs.