mit-aera/FlightGoggles #108: gate_locations argument is ignrored by renderer

lukked, 2019-03-14:
Whether I run this:
roslaunch flightgoggles scorer.launch level:=final gate_locations:=0 use_external_renderer:=1
or this:
roslaunch flightgoggles scorer.launch level:=final gate_locations:=1 use_external_renderer:=1

The /uav/GateX/location parameters change, but the gates on the rendered images are always at the exact same locations. 
Am I misunderstanding something? Please give me a hint. Thanks.

varunmurali1, 2019-03-14:
The `use_external_renderer:=1` is the issue. The gate locations are not being reloaded as they are passed as arguments to the renderer on startup. Please remove `use_external_renderer:=1` and try again.

lukked, 2019-03-14:
Removing use_external_renderer:=1 didn't help. 
But now, as soon as rviz starts, the renderer throws a bunch of error messages that usually indicate a crash and ending with:

`
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
`

But, oddly, the renderer appears to be working after that.

Gate positions remain remain untouched. At least Gate 10, that I am looking at.

If use_external_renderer:=1 is a problem, that must be reported to alphapilot as they want people to run their script with this option (and they will likely run it that way on AWS).

Winter-Guerra, 2019-03-14:
Hi @lukked, 

It looks like not all fixes listed in #80 were applied (removing `use_external_renderer` being one of the fixes). Could you try applying those fixes?  

Could you also try posting your player.log? 

I suspect that the renderer is looking for the perturbation file on startup, but is looking in the wrong place. Are the perturbation files installed to `flightgoggles/config/perturbations` ?

Best,
-Winter

lukked, 2019-03-14:
Oh, something's gone badly here.
beginning of Player log
[startofPlayer.log](https://github.com/mit-fast/FlightGoggles/files/2966869/startofPlayer.log)

then 38MB of binary crap...
then end of Player.log
[endofPlayer.log](https://github.com/mit-fast/FlightGoggles/files/2966872/endofPlayer.log)

yamls are present in directory src/flightgoggles/flightgoggles/config/perturbations

i'm checking #80 now.

lukked, 2019-03-14:
#80:
that's only the 
uav namespace: check
seq 0 24: check
remove use_external_renderer:=1: check

I corrected that already before (except use_external_renderer), 
No progress.

lukked, 2019-03-14:
There are a couple of readable messages in the beginning part of Player.log:
[startofPlayer_.log](https://github.com/mit-fast/FlightGoggles/files/2967395/startofPlayer_.log)

Winter-Guerra, 2019-03-14:
Hi @lukked,

Are you running the FlightGoggles renderer separately? If so, then you will not see the gates move. It is important that you close all renderer nodes before trying to load a new environment configuration. 

I was able to verify that the challenge file perturbations correctly affect the environment using the following steps:

* Download [AlphaPilot leaderboard challenge file zip](https://www.herox.com/alphapilot/resource/319)
* Unzip challenge files zip.
* Apply patches from #80 to unzipped files.
* Use instructions in unzipped README_v2.md to install files
* Close all ROS nodes and FlightGoggles renderers.
* Run layout 0: `roslaunch flightgoggles scorer.launch level:=final gate_locations:=0`
* Run layout 1: `roslaunch flightgoggles scorer.launch level:=final gate_locations:=1`

In layout 0/1 I was able to verify that gate 10 moved horizontally by about a meter or so in `v2.0.3` of FlightGoggles.

You could also try manually setting the FlightGoggles environment perturbation using `rosrun flightgoggles FlightGoggles.x86_64 -obstacle-perturbation-file /path/to/perturbations/perturbed_gates_0.yaml`

Best,
-Winter

lukked, 2019-03-14:
Hi @Winter-Guerra,
you correctly guessed the major part of the problem.
There was a render window left open somewhere in the background, and ros was obviously confused about that. That likely caused the corrupted Player.log.
My fault, I'm really sorry about that.

But still there must be some other bug.
with
roslaunch flightgoggles scorer.launch level:=final gate_locations:=0 1 2...
The renderer sometimes crashes.
I can't reproduce when this happens, but here is a Player.log after a crash:
[Player_crash.log](https://github.com/mit-fast/FlightGoggles/files/2968120/Player_crash.log)

and here after a working session:
[Player_ok1.log](https://github.com/mit-fast/FlightGoggles/files/2968131/Player_ok1.log)
last messages repeated many more times original file is 1.2MB

Thanks a lot.

lukked, 2019-03-14:
Solved, now  a crashed rospy node remained active in a terminal and caused the trouble. 
Sorry about the confusion I caused.

varunmurali1, 2019-03-14:
No worries. Glad it works.

closed 2019-03-14 by varunmurali1