mit-aera/FlightGoggles #110: Stereo images correctly rendered?

lukked, 2019-03-15:
I am looking at the scorer.launch initial position view in stereo. It appears to me that the right image is merely a zoomed out version of the left image. 
Could you please check?
Compare the the left and right edges if the images.

lukked, 2019-03-15:
here is what I mean:
[image]

Winter-Guerra, 2019-03-15:
Hi @lukked,

If you run the stereo disparity example, you can verify that stereo is working by viewing the reconstruction results. The left and right camera definitely have different views. I believe that what you're seeing is the result of an optical illusion. 

Best,
-Winter

lukked, 2019-03-15:
Hi @Winter-Guerra ,
ok I'll look at that example.
But on the image above right either 
1) has a wider field of view or 
2) it is taken from a position behind left. 
Camera axis is the same.
Right is supposed to be translated horizontally.

lukked, 2019-03-15:
Image center (coordinates 1024/2,768/2) are in both images **exactly** at the same position: at lower left corner of gate 10.
You can check this by counting pixels. That means that the camera axes are identical.

lukked, 2019-03-15:
Here is a possible explanation: 
if ros and unity don't use the same coordinate system somebody might have simply translated the right camera back instead of right.

Winter-Guerra, 2019-03-15:
Hi @lukked,

Thanks for raising this issue! I've just pushed a patch that fixes this issue. While merging in this feature, an issue with TF lookup got introduced into the codebase. Here's what the stereo reconstruction looks like now. I'm terribly sorry about that...

Best,
-Winter

[image]

commit 8e93faad (2019-03-15): Fixed TF tree lookup issue for stereo images. Fixes #110.
closed 2019-03-15 by Winter-Guerra