mit-aera/FlightGoggles #111: BUIld_INSTALL_FAILURE: python_distutil_install.sh returns error code:

SimpleUAVs, 2019-03-16:
Build install fails with following error.
[fg_build_err1.txt](https://github.com/mit-fast/FlightGoggles/files/2973299/fg_build_err1.txt)

Winter-Guerra, 2019-03-16:
Hi Jeff Zika,

`catkin build` does not trigger this issue for us on master. Additionally, the folder in question `flightgoggles_reporter` has not been changed in [8 days](https://github.com/mit-fast/FlightGoggles/commit/ca00b61b579f73cfd2420fb55d579884a88f6965). What is the specific command that you're using to produce this issue?

Best,
-Winter

SimpleUAVs, 2019-03-16:
Catkin build has never worked for me, so we use catkin_make then catkin_make install.

Winter-Guerra, 2019-03-16:
Hi Jeff,

The `catkin_make` toolchain is outdated and has been largely replaced with `catkin build` provided by the [`catkin_tools`](https://catkin-tools.readthedocs.io/en/latest/index.html) package. As such, we don't officially support `catkin_make install`. However, I've pushed a set of changes to [`feature/catkin_make_support`](https://github.com/mit-fast/FlightGoggles/tree/feature/catkin_make_support) that should fix the error that you're seeing. Please let me know if this works for you!

Best,
-Winter

SimpleUAVs, 2019-03-16:
Thanks, but still there in a different form. Error text attached.

[build_break-03-16-2019.txt](https://github.com/mit-fast/FlightGoggles/files/2974477/build_break-03-16-2019.txt)

Winter-Guerra, 2019-03-16:
This error message suggests about the catkin is trying to build using old cached build files. Could you please try:
```
cd ~/catkin_ws/
catkin_make clean
cd src/flightgoggles
git checkout origin/feature/catkin_make_support
cd ~/catkin_ws/
catkin_make
```

SimpleUAVs, 2019-03-18:
Build worked. Core.launch worked. Shutdown did not. Require SIGKILL of Unity.

Winter-Guerra, 2019-03-18:
I'm glad to hear that the build worked! The SIGKILL is a different issue. ROSlaunch handles this signal escalation automatically.

closed 2019-03-18 by Winter-Guerra

Winter-Guerra, 2019-03-19:
FYI, this branch has been merged into master.