mit-aera/FlightGoggles #113: BUILD_BREAK: TF transform error. Flightgoggles unusable

SimpleUAVs, 2019-03-16:
Running flightgoggles from core.launch (and every other launch file) generates the following error:

"Could NOT find transform for /uav/camera/left/nwu in time: "uav/camera/left/nwu" passed to lookupTransform argument source_frame does not exist"

Winter-Guerra, 2019-03-16:
Hi Jeff Zika,

[This commit](https://github.com/mit-fast/FlightGoggles/commit/2f33eaea1c6ac2cfc9c3f051280a8a4141c5b3b5) should have fixed the issue that your seeing. Could you please do another git pull? 

Thanks,
-Winter

SimpleUAVs, 2019-03-16:
Ok, seems stable for now. Please keep this open until soaked.

SimpleUAVs, 2019-03-16:
Still there, just runs a LOT slower now. Problem is with the last commit to flightgoggle_ros_bridge with fix to bug #110.

 

jz

SimpleUAVs, 2019-03-16:
Error message is now:

"Could NOT find transform for /UAV/camera/left/nwu in time: Lookup would require extrapolation into the future. Requested time 58.057310245 but the latest data is at time 58.056268578, when looking up transform from frame [/uav/camera/left/nwu] to fram [world]"

SimpleUAVs, 2019-03-16:
still there.

Winter-Guerra, 2019-03-16:
Hi Jeff,

The TF lookup issue that you're seeing is a new issue caused by the fix to #110. It should now be patched in commit d48e586. 

Thanks,
-Winter

SimpleUAVs, 2019-03-16:
Seems to work now. Thanks

 

jz

commit d48e586d (2019-03-16): Use last TF in buffer for render request instead of waiting for next TF. Resolves #113.
closed 2019-03-16 by Winter-Guerra