mit-aera/FlightGoggles #114: reporter reports drone collision when starting scorer.launch

mgandhi635, 2019-03-16:
Hello,

After the latest pull, there are instances where the simulator reports a collision before the drone even starts moving, or in the midst of an open region during flight (where there are clearly no obstacles.) I'll upload a video to demonstrate, but let me know what information I can provide to help reproduce and solve this bug.

Thanks!

System info:
Ubuntu 16.04, ROS kinetic

mgandhi635, 2019-03-16:
Okay, I saw the thread on #113 , and reverted to the commit before that (d01565636157298baa75bba85fe593e51e0f1417) and I am no longer having this issue.

Winter-Guerra, 2019-03-16:
Hi @mgandhi635,

None of the most recent changes introduced in #110 and #113 should change how collisions are detected or reported. We are unable to reproduce the symptoms you describe. Could you please try again with the latest commit on master?

Best,
-Winter

mgandhi635, 2019-03-16:
I tried again with the latest commit on master, and it seemed to have resolved the issue. I'll let yall know I encounter it again, or I can reproduce it reliably.

Thanks!

Winter-Guerra, 2019-03-16:
Awesome! Closing the issue for now. Please feel free to reopen it if needed.

roythompson, 2019-03-22:
We are seeing this same issue with the latest version from master.

roythompson, 2019-03-22:
To clarify, when we see the collision it is almost always around 4-8 seconds after the reporter starts; however we don't see any control commands until around 12 seconds.

Winter-Guerra, 2019-03-22:
Hi @roythompson,

Do you happen to have a ROSbag of the issue? Is this issue occurring after a previous run has been terminated via collision? What's the STDOUT of the roslaunch terminal when this issue occurs?

Thanks,
-Winter

roythompson, 2019-03-22:
This happens at the start of a fresh run.  On the terminal I see:

`[INFO] [1553230389.906855, 8.179169]: The drone collided!`

I also captured a ROSbag for this.  There is 1 `/uav/collision` message and 0 `/uav/input/rateThrust` messages.  Please let me know the best way to send this to you or if there is a filtered version of it I can capture (it ran for 5 seconds on all topics and is 100 MB).

Winter-Guerra, 2019-03-22:
Hi @roythompson,

Thanks for the info! If you have time, could you please filter the bag such that `/uav/camera/*` topics are removed? Then it should be small enough to post here.

In the meantime we'll work on a small patch to the reporter. The reporter should ignore all collisions before the drone is armed.

Best,
-Winter

tristandeborde, 2019-03-22:
Hi @Winter-Guerra  , 
We are also encountering the same problem with the latest commit from master. 
Did you manage to identify a cause for this bug ? 
Thanks in advance for your help

Winter-Guerra, 2019-03-22:
Hi all,

We think that the behavior you have been seeing may be due to previously cached collision messages. Commit 80b1a48492d78f078b6bd67ec0275be9faa62c63 adds a check where the reporter does not consider collisions until the drone is armed. Please note that this is a rare edge case, the general behavior of the reporter should not be affected at all.

closed 2019-03-22 by Winter-Guerra