mit-aera/FlightGoggles #115: Lots of noise on the ground truth TF

sneheshs, 2019-03-17:
We are seeing a massive error in the ground truth of the tf. Is this by design? Due to this issue, we are having issues with validating our position controllers. Our bearing plots of a completely stationary drone looks like the screenshots attached here. We just spent 12 hours debugging and trying to get around this. Just noticed folks saying there are tf errors. Can you please clarify what you folks are seeing? My question is WHY is there so much noise (or any noise at all) in the ground truth? The noise looks artificially injected with random intervals with some periodic function offsets. Appreciate any help or direction in this matter.

[image]

[image]

Winter-Guerra, 2019-03-18:
Hi @sneheshs,

The ground truth tf, i.e. `/uav/imu`, corresponds to the true vehicle position and does not contain any measurement noise. This can be queried simply using [tf listener](http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20listener%20%28C%2B%2B%29) from "world" to "/uav/imu"

Could you maybe clarify exactly what your plots are? What are you plotting exactly and what is the quantity on the x-axis?

Regarding noise in the simulation: There are two noise sources.

1) Measurement noise is added to the accelerometer and gyro measurements available in the IMU topic. The noise variance is added to the message as per message spec, see the wiki. Its value can also be adjusted in the parameter server. The gyro measurement noise also enters into the dynamics through the angular rate controller.

2) Process noise is added in the form of additive vehicle linear and angular accelerations. The corresponding parameters are also changeable in the parameter server. The process noise is added to simulate behaviors that also occur in real flight, such as turbulence and mechanical inbalances.

Attached are two plots that show the ground truth tf for a simulation run. The data was obtained using the teleop example. After a short thrust input to arm the drone, no further keyboard commands were given. The plots were made straight from a rosbag without any post processing, except the quaternion to Euler conversion. As you can see, the drone drifts from the initial position, as expected due to the measurement and process noises. Position and attitude are continuous.

[image]

[image]

Hope this helps

sneheshs, 2019-03-18:
Hi Winter-Guerra,

Thanks! Yes, this is extremely helpful. The reason we were seeing the noise was that we were not filtering ROS messages by the frame id so we were using both world (/uav/imu) as well as odom. I am wondering why FG didn't publish them as two different topics rather combine them into one -- I guess with the tf listerning you can easily filter if we already knew that this was the case. We did not notice this in the documentation; we must have overlooked this part.

Best,
Snehesh

closed 2019-03-18 by eatal