mit-aera/FlightGoggles #116: Stereo baseline?

lukked, 2019-03-18:
I think stereo baseline of rendered images is around 15cm. Could you please verify?

Winter-Guerra, 2019-03-18:
Hi @lukked,

The stereo baseline is 32cm and is defined in [these transforms](https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/launch/core.launch#L57). Please make sure you have the latest version of the codebase (git pull) if you're having issues with the stereo baseline.

lukked, 2019-03-18:
I have the latest version,
So, you are sure that unity actually renders with these parameters?

lukked, 2019-03-18:
wait, my version is d48e586.
i'll check the last commit now

lukked, 2019-03-18:
wrong, I have the last version 201af8d1.

Winter-Guerra, 2019-03-18:
What exactly is the issue that you are seeing? Do you believe the scale of the reconstruction is incorrect?

lukked, 2019-03-18:
I'm able to reconstruct correctly if I assume a 15cm baseline.

Winter-Guerra, 2019-03-18:
Hi @lukked,

Thanks for pointing out this issue! We are able to reproduce a scale inconsistency in the reprojected points using the stereo example. Setting the baseline to 0.16 m seems to make the reprojected points consistent with the movement of the drone in global frame. We're investigating where this issue is coming from and will update you asap.

Best,
-Winter

Winter-Guerra, 2019-03-19:
Hi @lukked,

We've traced down the root cause of the scaling issue to [this line in the renderer.](https://github.com/mit-fast/FlightGogglesRenderer/blob/master/FlightGoggles/Scripts/CameraController.cs#L506) We've pushed a hotfix to the ROS bridge node that alleviates the symptoms caused by this errant line of code.

*Important notes:*
- The baseline of the stereo pair is 32cm (no change).
- Similar to the monocular camera case, the left camera of the stereo pair is in *center of mass of the drone*.
- We also tweaked the example settings for reconstruction to improve reconstruction accuracy.

Here is a [video](https://youtu.be/hIdns3Z82c8) of this for reference.

Best,
-Winter & the MIT FlightGoggles Team

lukked, 2019-03-19:
Looks good now.
Thanks!

varunmurali1, 2019-03-19:
Glad it works for you!

closed 2019-03-19 by varunmurali1