mit-aera/FlightGoggles #117: Can't run glxgears on AWS P3 Instance. Could not open display :0

BhavyanshM, 2019-03-19:
Hello,

I have almost everything setup on AWS P3 instance using the AMI recommended on the FG-Wiki instructions for setting up the FG on AWS. The only problem I have been facing is with the X configuration. On running ```sudo X :0 &``` the output is ```$ (EE) 
Fatal server error:
(EE) Server is already active for display 0
	If this server is no longer running, remove /tmp/.X0-lock
	and start again.
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE)```

I would really appreciate any help at all in getting this fixed. Just don't have enough experience working with X servers. 

Thanks!
Bhavyansh

Winter-Guerra, 2019-03-19:
Hi @BhavyanshM,

It sounds like X is already running. You should only need to start X after every boot. If you're unsure as to the status of the system, I'd recommend doing a reboot and then `export DISPLAY=:0 && sudo X :0 &`.

Would you mind posting your `/etc/X11/xorg.conf`? Your virtual display might not be setup correctly. The full error for glxgears would also be helpful.

BhavyanshM, 2019-03-19:
> Hi @BhavyanshM,
>
> It sounds like X is already running. You should only need to start X after every boot. If you're unsure as to the status of the system, I'd recommend doing a reboot and then `export [...]

Hi @Winter-Guerra ,

On executing ```export DISPLAY=:0 && sudo X :0 &``` exactly after reboot is right when I see the output saying "Server is already active for display :0", However, here is the ```/etc/X11/xorg.conf```:
```
~$ cat /etc/X11/xorg.conf
# nvidia-xconfig: X configuration file generated by nvidia-xconfig
# nvidia-xconfig:  version 410.104

Section "ServerLayout"
    Identifier     "Layout0"
    Screen      0  "Screen0"
    InputDevice    "Keyboard0" "CoreKeyboard"
    InputDevice    "Mouse0" "CorePointer"
EndSection

Section "Files"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Mouse0"
    Driver         "mouse"
    Option         "Protocol" "auto"
    Option         "Device" "/dev/psaux"
    Option         "Emulate3Buttons" "no"
    Option         "ZAxisMapping" "4 5"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Keyboard0"
    Driver         "kbd"
EndSection

Section "Monitor"
    Identifier     "Monitor0"
    VendorName     "Unknown"
    ModelName      "Unknown"
    HorizSync       28.0 - 33.0
    VertRefresh     43.0 - 72.0
    Option         "DPMS"
EndSection

Section "Device"
    Identifier     "Device0"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
    BoardName      "Tesla V100-SXM2-16GB"
    BusID          "PCI:0:30:0"
EndSection

Section "Screen"
    Identifier     "Screen0"
    Device         "Device0"
    Monitor        "Monitor0"
    DefaultDepth    24
    Option         "AllowEmptyInitialConfiguration" "True"
    SubSection     "Display"
        Virtual     3200 1800
        Depth       24
    EndSubSection
EndSection
```

Cpruce, 2019-03-19:
`sudo service lightdm stop` to kill the X server already running

BhavyanshM, 2019-03-19:
> `sudo service lightdm stop` to kill the X server already running

Oh thanks @Cpruce , this actually did make ```glxgears``` run successfully. Should be able to launch FG now.  Thanks @Winter-Guerra and @Cpruce !

closed 2019-03-19 by BhavyanshM