mit-aera/FlightGoggles #118: flightgogglesRenderer crashes on P3 instance: mono_gdb_renderer_native_backtraces not supported on this platform

BhavyanshM, 2019-03-19:
Hi @Winter-Guerra 

I was able to fix the ```glxgears``` issue with X server. However, now on running the ```core.launch```, the flightgogglesRenderer node keeps terminating and restarting with this error in an infinite loop:
```
Debug info from gdb:

mono_gdb_render_native_backtraces not supported on this platform, unable to find gdb or lldb

=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

[uav/flightgogglesRenderer-3] process has died [pid 6906, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/flightgoggles/FlightGoggles.x86_64 -screen-quality Fastest __name:=flightgogglesRenderer __log:=/home/ubuntu/.ros/log/192b8c2a-4a20-11e9-9dff-0ea8d61e9652/uav-flightgogglesRenderer-3.log].
log file: /home/ubuntu/.ros/log/192b8c2a-4a20-11e9-9dff-0ea8d61e9652/uav-flightgogglesRenderer-3*.log
[uav/flightgogglesRenderer-3] restarting process
process[uav/flightgogglesRenderer-3]: started with pid [7240]
```

I would really appreciate any help. I do understand this is crunch time for everyone, and extremely appreciate your patience [FG team]

Winter-Guerra, 2019-03-19:
Could you please install gdb and also post the player.log?

Winter-Guerra, 2019-03-19:
Might also want to try to install `monodevelop-debugger-gdb ` package. Probably not necessary though, so don't spend too much time on it.

BhavyanshM, 2019-03-20:
Oh thanks yet again @Winter-Guerra Installing ```gdb``` fixed the issue right away

closed 2019-03-20 by BhavyanshM