mit-aera/FlightGoggles #119: IR beacon correspondence

LukeDitria, 2019-03-19:
Hi, we've been using the IR Beacons and our position etc to locate the gate corners in world coordinates and have been using the ground truth gate corner positions to verify our estimates. For some reason we could only get good estimates on gates that have not been perturbed from the nominal location. On further inspection all of our estimates using the ir beacons where essentially perfect matches to the nominal location even with gates that had been obviously perturbed from their nominal location.
Are the IR Beacons only given according to the nominal location, or are we just doing something very wrong?

Winter-Guerra, 2019-03-19:
Does the debug view in Rviz of the IR beacons on the RGB image look ok? If not, could you please send us a picture of the issue? Are you using v3 of the AlphaPilot challenge files?

LukeDitria, 2019-03-19:
Hi, thanks for your reply, 
we realized that our problem was the fact that the reporter.launch file doesn't pass core.launch the gate_locations arg so the renderer was putting the gates in the same nominal location, we though it was enough to change the gate_locations.yaml and challenge.yaml that was loaded to the parameter server.
thanks for your time

closed 2019-03-19 by Winter-Guerra