mit-aera/FlightGoggles #121: Running Flight Goggles on AWS g3s.x instance

madunclejack, 2019-03-20:
Hello, I know there have been previous issues opened with similar problems, I'm hoping this is both different enough and close enough to solving to be worth your time.

I'm running an AWS g3s.xlarge instance, with a Tesla M60 GPU. Since I'm running it through AWS, I did a lot of poking around to find a suitable VNC service that would properly forward X windows created by ROS, and landed on VirtualGL and TurboVNC. My client machine is an iMac, but I don't think that is factoring in right now. Some posts online suggest the problem may be a lack of VRAM. Running lspci -v -s on the GPU seems to say I only have 256MB of memory, but online specs say an M60 has 4GB.

I have nvidia drivers 410.104 downloaded from online and Vulkan API version 1.0.61. It is Ubuntu 16.04.

I installed Flight Goggles and ROS as per the wiki instructions, and I can't get roslaunch flightgoggles core.launch to run. It sends the error: 

Xlib:  extension "NV-GLX" missing on display ":1".
[uav/flightgogglesRenderer-3] process has died [pid 8121, exit code 1, cmd /home/awsgui/catkin_ws/devel/lib/flightgoggles/FlightGoggles.x86_64 -screen-quality Fastest __name:=flightgogglesRenderer __log:=/home/awsgui/.ros/log/1819934a-4ac5-11e9-b742-12e50dc0c734/uav-flightgogglesRenderer-3.log].
log file: /home/awsgui/.ros/log/1819934a-4ac5-11e9-b742-12e50dc0c734/uav-flightgogglesRenderer-3*.log

The VNC display is on :1. I'm attaching the uav-flightgogglesRenderer-3*.log and the Player.log. Please let me know if there is any other info I can provide to help, and I greatly appreciate your assistance.

[uav-flightgogglesRenderer-3-stdout.log](https://github.com/mit-fast/FlightGoggles/files/2986566/uav-flightgogglesRenderer-3-stdout.log)
[Player.log](https://github.com/mit-fast/FlightGoggles/files/2986567/Player.log)

Other things to note, glxgears works but vulkan-smoketest gives the same Xlib: extension error as above and doesn't run. And as far as I can tell, nvidia-intel drivers are not installed.

Winter-Guerra, 2019-03-20:
Hi @madunclejack ,

It looks like Vulkan is not able to find your render device. This could happen if your NVIdia driver or Vulkan driver is not installed. Is `libvulkan1` installed? What does `vulkaninfo` output? What is the content of `/etc/X11/xorg.conf`?

madunclejack, 2019-03-20:
Hello @Winter-Guerra 

It looks like libvulkan1 is installed, with version 1.0.61.1+dfsg1-1ubuntu1~16.04.1

The output of vulkaninfo is
===========
VULKAN INFO
===========

Vulkan API Version: 1.0.61

INFO: [loader] Code 0 : Found ICD manifest file /etc/vulkan/icd.d/nvidia_icd.json, version "1.0.0"
Xlib:  extension "NV-GLX" missing on display ":1".

Instance Extensions:
====================
Instance Extensions	count = 13
	VK_KHR_display                      : extension revision 21
	VK_KHR_external_fence_capabilities  : extension revision  1
	VK_KHR_external_memory_capabilities : extension revision  1
	VK_KHR_external_semaphore_capabilities: extension revision  1
	VK_KHR_get_physical_device_properties2: extension revision  1
	VK_KHR_get_surface_capabilities2    : extension revision  1
	VK_KHR_surface                      : extension revision 25
	VK_KHR_xcb_surface                  : extension revision  6
	VK_KHR_xlib_surface                 : extension revision  6
	VK_EXT_acquire_xlib_display         : extension revision  1
	VK_EXT_debug_report                 : extension revision  9
	VK_EXT_direct_mode_display          : extension revision  1
	VK_EXT_display_surface_counter      : extension revision  1
/build/vulkan-YO5iw0/vulkan-1.0.61.1+dfsg1/demos/vulkaninfo.c:1722: failed with VK_ERROR_INITIALIZATION_FAILED

Here's the xorg.conf
[xorg.conf.txt](https://github.com/mit-fast/FlightGoggles/files/2989948/xorg.conf.txt)

Also, I tried running roslaunch flightgoggles core.launch again for fun, and got an rviz crash after I jammed on Ctrl-C. It keeps trying to launch a window, but immediately closes it and posts the error from before. Here's the Ubuntu crash report in beautiful text form...
[rvizCrash.txt](https://github.com/mit-fast/FlightGoggles/files/2989984/rvizCrash.txt)

madunclejack, 2019-03-20:
I tried this in 3 different ssh terminals:
terminal 1: 'export DISPLAY=:0`
                   'sudo X :0`

terminal 2: 'export DISPLAY=:0`
                   'rosrun rviz rviz`

terminal 3: 'sudo rm -rf /tmp.X*`
                   '/opt/TurboVNC/bin/vncserver -geometry 1920x1080`

Without terminal 3, rosrun rviz seems to start, but if I try to launch a VNC host, it goes to display 1.

I rosrun rviz runs without printing errors, but flight goggles does the same thing as before. Also, the VNC client has a black screen, but that's probably my problem more than yours.

I tried the same setup with 'export DISPLAY=:1` and that also was stable for rosrun rviz, but not for flight goggles

Should flight goggles be able to launch on display 1? If so, some other posts are suggesting I didn't set up VirtualGL correctly.

Winter-Guerra, 2019-03-21:
Hi @madunclejack,

Hmm. This looks like a problem regarding VNC and GPU level access from within the VNC session. The recommended method of using FlightGoggles does not use VNC, but instead relies upon ROS_MASTER_URI tricks for forwarding debug information to the local client.

I would recommend the following:
- Try running FlightGoggles on `Display=:0`. Does FlightGoggles run there? If so, I'd try to keep FlightGoggles on `Display=:0` and run rviz on the VNC display. The `sudo rm -rf /tmp.X*` you are running might be transferring FlightGoggles to the VNC display.
- [This post](https://stackoverflow.com/questions/34805794/virtualgl-and-turbovnc-extension-glx-missing-on-display-0-0) seems to have helped some people in the past whom have had similar errors with other GPU-enabled programs. Additionally, [this guide](https://carla.readthedocs.io/en/latest/carla_headless/) might also help.

Best,
-Winter

Winter-Guerra, 2019-04-02:
Closing issue due to inactivity.

closed 2019-04-02 by Winter-Guerra