mit-aera/FlightGoggles #127: reporter.py not terminating on collision?

lukked, 2019-03-30:
I noticed that collisions only cause FG to reset the drone to initial position but all nodes remain active.
for example
roslaunch flightgoggles scorer.launch level:=final gate_locations:=0

does not terminate on collision but it used to, if I remember correctly this used to be different.
Now reporter.py stays active and consequently roslaunch is also not terminated, only the positioni reset.

Do you have any explanation for this? Is there any parameter that must be set?

Winter-Guerra, 2019-03-30:
We have not seen this problem before. Does this happen every time? If so, could you please send us a rosbag and perhaps a dump of your Ros params?

lukked, 2019-03-30:
It happens every time, no matter if I do rostopic pub -1 /uav/collision std_msgs/Empty or a real collision occurs.

This is displayed at start:
[ERROR] [1553959759.306791, 0.073958]: Challenge name could not be read

Gate positions do visually seem to change.

I'm actually trying to debug my submission to alphapilot which appears to crash immediately upon start of alphapilot's testing.

I tried:
rosrun flightgoggles scorer.sh
on my computer and stumbled upon this issue. 
I'm not sure if the submission issue is connected to the weird reporter.py behavior.

I can't prepare a rosbag right now but here are the dumps for
roslaunch flightgoggles scorer.launch level:=final gate_locations:=0 and
roslaunch flightgoggles scorer.launch level:=final gate_locations:=1
[dump0.yaml.txt](https://github.com/mit-fast/FlightGoggles/files/3025713/dump0.yaml.txt)
[dump1.yaml.txt](https://github.com/mit-fast/FlightGoggles/files/3025715/dump1.yaml.txt)

I case this helps: 
I have a ridiculously slow PCI bandwidth and a 1050Ti eGPU. Automatic clock scaling doesn't reliably work for me (typical symptom: multiple renderer crashes and then suddenly it works)
If I set manual clock scaling to 0.1 crashes are rare, topic rate is between 59 and 61 Hz so I guess that should be ok.

Do you think that the speed difference between my box and AWS could cause my controller to fail on AWS inspite of clock scaling?
I hope alphapilot doesn't do something nasty like secretly changing mass or so, because. being in hurry, I hardcoded that.

varunmurali1, 2019-03-30:
Those parameter dumps do not appear to have all the parameters. For instance I do not see a `/uav/challenge_name`. I also think that you possibly have an older version of these files from alphapilot. Could you please check if the following lines in your `scorer.launch` are in the right namespace and present in your `scorer.launch`? 

```
<group ns="/uav"> 
    <rosparam command="load" file="$(find flightgoggles)/config/challenges/gate_locations_$(arg gate_locations).yaml"/>
    <rosparam command="load" file="$(find flightgoggles)/config/challenges/challenge_final.yaml"/>  
  </group>
```

If they are in the right namespace could you please post your `scorer.launch`?

lukked, 2019-03-30:
Regarding challenge name: I don't know what's happening. There was this error message that I posted above.

The files below are from a collision using teleop with the ceiling right above the initial position at:
roslaunch flightgoggles scorer.launch level:=final gate_locations:=0

The bagfile should cover data shortly before and after collision.
Image related topics are not included.

[scorer.launch.txt](https://github.com/mit-fast/FlightGoggles/files/3025962/scorer.launch.txt)
[subset.bag.zip](https://github.com/mit-fast/FlightGoggles/files/3025967/subset.bag.zip)
[dump_.yaml.txt](https://github.com/mit-fast/FlightGoggles/files/3025968/dump_.yaml.txt)

varunmurali1, 2019-03-30:
I am not entirely sure how your params do not contain `/uav/challenge_name` when other parameters appear to be loaded. Perhaps, could you please check that the `challenge.yaml` is the same as the one provided by alphapilot and contains the `challenge_name`. 

Looking at this [line](https://github.com/mit-fast/FlightGoggles/blob/80b1a48492d78f078b6bd67ec0275be9faa62c63/flightgoggles_reporter/src/reporter.py#L98), the expected behavior is that if the challenge name is not defined the node signals shutdown. If this isn't the case, i.e. reporter does not shutdown at this error perhaps the shutdown signal is not being handled. In any case, the missing parameter would stop the reporter from running and logging any events.

lukked, 2019-03-30:
That's it. In a hurry I obviously I overlooked that they changed challenge_final.yaml.
Thanks a lot.

closed 2019-03-31 by Winter-Guerra

lukked, 2019-04-02:
Hi @Winter-Guerra,
the problem with reporter.py not terminating when challenge_name is missing is due to calling signal_shutdown() from outside the while loop. In addition, this causes the node to ignore subsequent calls to signal_shutdown() in the callbacks (like on collision). Same applies if the timeout parameter is missing. So the the name and timeout checks should be moved inside the while loop. I can make a pull request if you like.