mit-aera/FlightGoggles #131: Cannot create Vulkan instance on Ubuntu-16.04 host with nvidia-418 driver

edowson, 2019-04-13:
I am encountering an issue with getting the vulkan drivers to work properly on a new Ubuntu-16.04 host with nvidia-418 drivers:

```
$ vulkaninfo
===========
VULKAN INFO
===========

Vulkan Instance Version: 1.1.73

ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Attempt to retrieve either 'vkGetInstanceProcAddr' or 'vk_icdGetInstanceProcAddr' from ICD libGL.so.1 failed.
Cannot create Vulkan instance.
/build/vulkan-ZsgAIG/vulkan-1.1.73+dfsg/demos/vulkaninfo.c:812: failed with VK_ERROR_INCOMPATIBLE_DRIVER
```

edowson, 2019-04-13:
Edit `/usr/share/vulkan/icd.d/nvidia_icd.json` and rename `libGL.so.1` to `libGLX_nvidia.so.0`
```bash
sudo nano /usr/share/vulkan/icd.d/nvidia_icd.json
```

```json
{
    "file_format_version" : "1.0.0",
    "ICD": {
        "library_path": "libGLX_nvidia.so.0",
        "api_version" : "1.1.95"
    }
}
```

Ref:
[Vulkan does not work [closed] - Ask Ubuntu](https://askubuntu.com/questions/967335/vulkan-does-not-work/967910)

SimpleUAVs, 2019-04-13:
No.

edowson, 2019-04-14:
I was able to figure out how to get vulkan drivers to work from within an Ubuntu-16.04 NVIDIA GPU accelerated docker container with two Titan-V GPUs connected to the host system. 

The should should help fix issue #65.

In addition to creating the `/usr/share/vulkan/icd.d/nvidia_icd.json` file, you'll also need to download the nvidia drivers, extract the *.run package and manually extract the `libnvidia-glvkspirv.so.418.56` file to `/usr/lib/x86_64-linux-gnu/`, inside the docker image.

This will allow `vulcaninfo` and `vulkan-smoketest` to run correctly.

Additionally, running `rosrun flightgoggles FlightGoggles.x86_64` pops up the render window.

```bash
wget http://us.download.nvidia.com/XFree86/Linux-x86_64/418.56/NVIDIA-Linux-x86_64-418.56.run
NVIDIA-Linux-x86_64-418.56.run --extract-only
sudo cp NVIDIA-Linux-x86_64-418.56/libnvidia-glvkspirv.so.418.56 /usr/lib/x86_64-linux-gnu/
```

closed 2019-04-14 by edowson