mit-aera/FlightGoggles #133: use_sim_time

engcang, 2019-04-28:
Hi

about 1 month ago, when i tested my algorithm with flightgoggles simulator,
with use_sim_time "true", always the simulator seems slower than real time.
so it was good for testing fast flight.

but suddenly today i found that, even with <param name="use_sim_time" value="true"/>,
the simulator at first moves as real time (faster) and then changes into as sim time (slower) while flying, about few secs after started. even the how slow it changes depends on each simulating.

I don't understand why simulator doesn't apply sim time right after it starts :)

SnShine, 2019-04-30:
Yeah we too noticed this while working on the AlphaPilot challenge. As a hack we just started our control nodes a second later and it worked.

But the bug in FlightGoggles exists nonetheless.

Winter-Guerra, 2019-07-17:
Hi all,

You're seeing this delayed adjustment to the sim-time ratio as a result of how FlightGoggles performs adaptive sim-time scaling. The FlightGoggles simulator changes the clockrate depending on how well the renderer is performing. At startup, FlightGoggles tries to run in realtime, but then quickly decreases the clockrate after it has a better idea of how fast your computer is able to run the renderer. As a workaround, I would recommend waiting a few seconds after startup for the clockrate and framerate to stabilize.

I'm going to close this issue as `#wontfix`, since this is expected behavior with a simple workaround. Thanks for the bug report!

closed 2019-07-17 by Winter-Guerra