mit-aera/FlightGoggles #138: Message passing delay when running FG built from source

paulerikf, 2019-09-23:
Hi there...

I'm experiencing a sizeable delay when using a version of FlightGoggles built from source in Unity.  It's ~5-10 seconds from when things happen in the dynamics node until they show up in the actual renderer, and about ~10 seconds until images / collisions etc. from the renderer show up in ROS again.

This does not happen with the downloadable binary you provide.  However it happens when running FlightGoggles in Unity, or when running a built version in Windows or Ubuntu.

Any ideas about what could be causing this, or how to fix it?

Thanks!

Winter-Guerra, 2019-09-23:
Hey Paul,

Thanks for the bug report. Could you try checking out [this commit](https://github.com/mit-fast/FlightGogglesRenderer/commit/b91e283824d189b104c06890783781140590bae9) of the Unity3D renderer source code and reporting back your results? Alternatively, you could uncomment lines 206 & 212 in [cameraController.cs](https://github.com/mit-fast/FlightGogglesRenderer/blob/master/FlightGoggles/Scripts/CameraController.cs#L206). I suspect that the commenting out of these lines in the latest commit might be causing your issue.

paulerikf, 2019-09-23:
Thanks Winter,

Uncommenting lines 206 & 212 helped a lot, and going back to the referenced commit worked even better. 

Actually, the commit didn't build because it couldn't find Newtonsoft.json on Line 31 - but using that commit's version of CameraController.cs worked great!

Thanks again!

Winter-Guerra, 2019-09-23:
Glad to hear it! I'll leave this issue open until I push a partial revert.

sabinach, 2019-10-03:
Hi Winter,

I am also have the same issue with the delay. I am running the Unity renderer on Windows, and FlightGoggles ROS on Ubuntu. I've uncommented lines 206 & 212, and made sure to included "transform" in internal_object_state.gameObj.transform.position from the commit in CameraController.cs. However, there is still the ~5-10 second delay before both Unity and ROS updates the images/collisions. 

Thanks for your help!

paulerikf, 2019-10-05:
Running the renderer in Windows has not worked great for me.  But after using the [mentioned commit](https://github.com/mit-fast/FlightGogglesRenderer/commit/b91e283824d189b104c06890783781140590bae9) the delay disappeared at least.  Do you have the same issue with and without `use_sim_time`?

sabinach, 2019-10-07:
Yes my code is using the commit with the "transform" added. Toggling use_sim_time in the launch file still causes the delay..

ThomasCarstens, 2020-02-22:
Hi Sabinach, did you find a solution? No difference here either toggling use_sim_time or changing clockscale...