mit-aera/FlightGoggles #145: Strange quaternion issue

juanmed, 2020-01-09:
Hello,

I used flightgoogles in the past for the AlphaPilot Challenge and tried it again this week for other simulations. I am having an issue which an strange error message I did not face before: 

```bash
[ERROR] [1578553914.709314675]: Ignoring transform for child_frame_id "uav/imu" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)

         at line 257 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "uav/imu" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan nan) (nan nan -nan -nan)
         at line 244 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "uav/imu" from authority "unknown_publisher" because of an invalid quaternion in the transform (nan nan -nan -nan)
         at line 257 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp

```

This simulator starts correctly using `roslaunch flightgoogles core.launch` but right at the moment I sent the arming message to the `/uav/input/arm` topic, the issue above gets printed repeatedly at a high rate. This issue does not happen all the time, sometimes I am able to fly (using the keyboard or by sending control messages to `/uav/input/rateThrust/`) the vehicle without issues.

I will appreciate any help you could provide. Thank you. 

My system is:
OS: Ubuntu 16.04 64bit
CPU: Intel® Core™ i5-8400 CPU @ 2.80GHz × 6 
Graphics: GeForce GTX 1050 Ti/PCIe/SSE2
RAM: 8Gb

eatal, 2020-01-09:
Hi @juanmed ,

I have not seen this issue myself. Can you make sure that all control messages on `/uav/input/rateThrust/` are in order? For example, can you recreate the issue while sending constant numeric values as commanded angular rates and thrust?

After that, a good starting point for debugging may be to add some diagnostics in the block starting [here](https://github.com/mit-fast/FlightGoggles/blob/ad28590e19c8b74171f65e973331e70bdd5358da/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp#L323) to find out the origin of the NaNs.

juanmed, 2020-01-14:
Thanks for you input. I will try out your suggestions and let you know.

eatal, 2020-02-07:
I'll close this for lack of new information. Feel free to reopen in case of any updates.

closed 2020-02-07 by eatal