mit-aera/FlightGoggles #147: New commits introduce regressions

varunagrawal, 2020-02-07:
Hi,

I recently updated my local version to the latest commit and the keyboard command not longer works.
The breaking change was introduced in [this](ff5fca7f1f359b1f749f3c3bf9046d0bfa0e6441) commit.

It would be great if this could be fixed.

varunagrawal, 2020-02-07:
@Winter-Guerra @eatal any updates?

eatal, 2020-02-07:
Hi @varunagrawal ,

Can you describe the issue you are experiencing in a bit more detail?

The commit you referenced should not affect the operation of the keyboard teleop node. Are you using the updated take-off procedure as summarized in the commit message as "the vehicle is armed by publishing a message to the topic `/uav/input/arm`" (see also in the FAQ [here](https://github.com/mit-fast/FlightGoggles/wiki/FAQ#why-is-the-vehicle-not-moving))?

varunagrawal, 2020-02-07:
Oh I see. I was using an older version that didn't require that parameter
to be set so this was unintuitive to me. Can you please add this to the
"Getting Started" instructions?

closed 2020-02-07 by varunagrawal