mit-aera/FlightGoggles #149: Potential Thrust mixing mistake

jellewestenberger, 2020-02-28:
Hi! 
I think I spotted an error in this thrust mixing method:
https://github.com/mit-fast/FlightGoggles/blob/ad28590e19c8b74171f65e973331e70bdd5358da/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp#L696-L702

Here, the motorspeeds corresponding to the body angular acceleration around the z-axis is calculated using the torque coefficient. But the torque coefficient only relates to the body angular acceleration through the propeller acceleration, not propeller speed! 

However, the aerodynamic moment contribution to the z-axis does depend on propeller speed.
Therefore, I think if  `torqueCoeff_` in this calculation is replaced by `aeroMomentCoefficient(2,2)`, you should get more realistic results.

I noticed that in an earlier version a related error was made here, since the propeller speed is used in calculating the moment around z:
https://github.com/mit-fast/FlightGoggles/blob/8a0af6cdcd4d5fa9cc002959b294186008e82dd2/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp#L323-L327

Which is actually corrected in multicopterDynamicsSim, but the thrust mixing scheme hasn't been updated accordingly.

Please correct me if i'm wrong.

jellewestenberger, 2020-02-28:
Actually, I made a mistake here.
 `aeroMomentCoefficient(2,2)` corresponds to the quadcopter's body aerodynamic dynamics, not the propeller aerodynamics. So it should not replace `torqueCoeff_`. But a coefficient for the propellers aerodynamic moments would be required.

eatal, 2020-02-28:
Hi Jelle,

thank you for your message.

To model the thrust and control moment we use a relatively standard expression, see for example equations (5), (6), and (7) in https://ieeexplore.ieee.org/document/8619621. (Note: The axis system in the paper differs from the one used in FlightGoggles, so some signs may be switched.) The model consists of aerodynamic contributions based on the motor speeds squared (using the thrust and torque coefficients), and inertial contributions based on the motor acceleration using the motor inertia. In this model, the torque coefficient represents the ratio between the moment around the z-axis and the motor speed squared. This naming is common, see for example https://web.mit.edu/16.unified/www/FALL/thermodynamics/notes/node86.html.

In the thrust mixing method you linked above, i.e. https://github.com/mit-fast/FlightGoggles/blob/ad28590e19c8b74171f65e973331e70bdd5358da/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp#L696-L702, the model is simplified to only include the aerodynamic contributions based on the thrust and torque coefficients. This way, the thrust mixing can be computed by a single matrix computation.

In multicopterDynamicsSim, the more elaborate model including both aerodynamic and inertial contributions is used [here](https://github.com/mit-fast/multicopterDynamicsSim/blob/241f96b7eee00a10c10d523aed401a6f01600532/multicopterDynamicsSim.cpp#L374). First the aerodynamic contribution is computed using the torque coefficient and the motor speed squared, and on the next line the inertial contribution is added using the motor inertia and the motor acceleration.

Hope this helps

Ezra

jellewestenberger, 2020-02-28:
Hi Ezra. 
Thanks for your quick response!

You have made it clear to me. 
It seems I had misinterpreted the unit of the torquecoefficient. 

Thank you for taking the time to explain it so elaborately!

closed 2020-02-28 by jellewestenberger

eatal, 2020-02-28:
You're welcome!