mit-aera/FlightGoggles #156: Using different scenes

engcang, 2020-07-07:
Hi. 

I emailed Winter about this question but I think he does not check his MIT email anymore after graduation.

I found there is a few more available scenes in the ROSClient.cpp code
and I re-built the package with the other scenes and found nothing come up after changed the scenes.

Is there any other available scenes for FlightGoggles? and for that, is there specific way to change the scene instead of simply change the flightGoggles.state.sceneFilnename?

Since I really love your work and use this simulator a lot, I would like to use the other maps if possible!

Thanks.

EungChang

varunmurali1, 2020-07-07:
Hey @engcang, with the current binary only one scene is usable. If you want scenes from the Blackbird dataset you can use an older release of the renderer. You should only have to change the flightGoggles.state.sceneFilname to use those scenes. 

We are trying to add more scenes to the current binary. Please stay tuned!

closed 2020-07-07 by varunmurali1

engcang, 2020-08-03:
> Hey @engcang, with the current binary only one scene is usable. If you want scenes from the Blackbird dataset you can use an older release of the renderer. You should only have to change the [...]

Thanks for reply @varunmurali1 .

I can use other scenes from the Blackbirrd dataset? that is really interesting.
Do I have to download whole dataset and then follow some specific steps to set configuration to use those scenes?
Or is everything going to be automatically set by just changing the flightGoggles.state.sceneFilename?

Thank you in advance!