mit-aera/FlightGoggles #157: Creating a custom vehicle for simulation in the loop testing

Mohit505Git, 2020-07-27:
Hey there,

I found the concept of flight goggles to be very amazing and enjoyed reading the paper on your work. Is it possible for one to design a custom configuration in which the multirotor is mounted with external structures or actuators which may interact with the ground of the environment? Any help will be truly appreciated

Best 
- Mohit Singh

varunmurali1, 2020-07-27:
We unfortunately do not model any interactions with the environment. The renderer and dynamics simulation are separated to allow them to run at different rates and preserve modularity for all of our applications. As such the dynamics simulation has no access to the environment model. However, it is possible to use the returned collision detections and modify the dynamics simulation node appropriately if you require only simplistic interaction.

Mohit505Git, 2020-08-01:
Thanks for the answer! 
In that case, does the environment have any impermeability condition on collision, or is it only collision detection?