mit-aera/FlightGoggles #159: Changing the camera extrinsic

AlessandroFornasier, 2020-09-17:
Hi, thanks a lot for open sourcing this amazing work, I am actually working on extending some functionalities of FG and I was wondering if it is possible to have different camera extrinsic (mainly a different rotation wrt imu) acting only on the ros bridge side.

Thanks a lot in advance for the answer.

varunmurali1, 2020-09-17:
Yes, it is easy to change the camera extrinsic. Take a look at `core.launch`. The TF for cameras is defined there in line 56 and 57. You could modify the transform directly from there since the ros bridge works by querying the tf ( see https://github.com/mit-fast/FlightGoggles/blob/f25e1c2555d0c83f3cbac441d4e6973f09e2d0b9/flightgoggles_ros_bridge/src/ROSClient/ROSClient.cpp#L190)

AlessandroFornasier, 2020-09-18:
Thanks for the answer, this was what I exactly tried without success before opening the issue, anyway, I figured out how to do that looking at the tf tree. I actually needed to change both lines 56-57 and 62-63 since the tf the ros bridge is querying is "world"--"uav/camera/left/internal_nwu".

Thanks a lot again for your fast reply.

closed 2020-09-18 by AlessandroFornasier