mit-aera/FlightGoggles #161: Issue on Installing Flightgoggles Simulation Framework on Local Machine

charyyev, 2021-03-24:
I followed all the steps on "Installing Flightgoggles on Local Machine", and when I run catkin build it couldn't build flightgoggles package. Here is the displayed error message.

Errors     << flightgoggles:make /home/sapar/catkin_ws/logs/flightgoggles/build.make.000.log
CMake Error at /usr/share/cmake-3.10/Modules/ExternalData.cmake:1121 (message):
  

  Object MD5=1ca07cb660744ac6425c00f030b32df0 not found at:

    http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/HashedReleases/MD5/1ca07cb660744ac6425c00f030b32df0 ("Couldn't resolve host name")
    ftp://fast-nas.mit.edu/Public/FlightGoggles/HashedReleases/MD5/1ca07cb660744ac6425c00f030b32df0 ("Couldn't resolve host name")

make[2]: *** [renderer_version/FlightGoggles_Data.tar.gz-hash-stamp] Error 1
make[2]: *** Waiting for unfinished jobs....
CMake Error at /usr/share/cmake-3.10/Modules/ExternalData.cmake:1121 (message):
  

  Object MD5=7031696e88517c28b8ddcf82760363b0 not found at:

    http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/HashedReleases/MD5/7031696e88517c28b8ddcf82760363b0 ("Couldn't resolve host name")
    ftp://fast-nas.mit.edu/Public/FlightGoggles/HashedReleases/MD5/7031696e88517c28b8ddcf82760363b0 ("Couldn't resolve host name")

make[2]: *** [renderer_version/FlightGoggles.x86_64-hash-stamp] Error 1
make[1]: *** [CMakeFiles/flightgogglesBinaryDownload.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

Does anyone know why this might be happening?

ThomasCarstens, 2021-03-30:
Same issue here, I found issue #23 deals with this, by downloading these files from the release pages. Please let me know if it works for you!

varunagrawal, 2021-04-05:
We shouldn't need to download the binaries separately. It looks like something changed on cloudfront on those changes have not been reflected in the CMake.

varunmurali1, 2021-04-16:
@varunagrawal this is an intentional and planned move away from CloudFront. In future, the cmake download workflow will no longer be supported. I updated the documentation to reflect this for now, and this will be reflected in code in the next big release.

closed 2021-04-16 by varunmurali1

varunagrawal, 2021-04-17:
Great, thanks for the update @varunmurali1.