mit-aera/FlightGoggles #165: FlightGogglesV3 binary is not working

MajdWardeh, 2021-06-07:
Hi,

I am familiar with this simulator since its release, and I've been tracking changes and new releases regularly.
I've downloaded the release for version 3 of FlightGoggles and gotten the following issue:
In the release folder, there is no file called ./FlightGoggles.x86_64 but there is FlightGogglesv3.x86_64. (just mentioned this to update your wiki if you want).
After running the binary by executing ./FlightGogglesv3.x86_64, a window appeared and closed directly.
I got the following output on the terminal:
```
Set current directory to /home/majd/Downloads/FlightGogglesv3-release
Found path: /home/majd/Downloads/FlightGogglesv3-release/FlightGogglesv3.x86_64
```
Here is the content of the Player.log file:

```
Desktop is 1280 x 720 @ 60 Hz
[Vulkan init] extensions: count=13
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2018.3.0f2 (6e9a27477296)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=13
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: count=89
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_imageless_framebuffer, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance3, enabled=0
[Vulkan init] extensions: name=VK_KHR_multiview, enabled=0
[... 95 lines cut ...]

Terminated ZMQ sockets.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 45)

Configuring sockets.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 45)

Sockets bound.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 45)

Fallback handler could not load library /home/majd/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32
Fallback handler could not load library /home/majd/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32.so
Fallback handler could not load library /home/majd/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/Kernel32
Fallback handler could not load library /home/majd/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32
Fallback handler could not load library /home/majd/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32.so
Fallback handler could not load library /home/majd/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1280 x 720 @ 60 Hz
InitializeOrResetSwapChain 1024x768 samples=1
InitializeOrResetSwapChain 1024x681 samples=1
```

varunmurali1, 2021-06-07:
The wiki is deprecated for v3.0+. It has not been deleted to maintain legacy documentation. As mentioned in the readme, the code documentation is [here](https://flightgoggles-documentation.scrollhelp.site/fg/). I will perhaps highlight this in the readme. 

Looking at your Player.log the image you are requesting is 1024x768 but your screen is only 1280x720. As you probably know, a limitation of FlightGoggles is the requirement of screensize being >= to the requested render size. Please try reducing your image size.

MajdWardeh, 2021-06-08:
Thank you so much.

closed 2021-06-23 by varunmurali1

MajdWardeh, 2021-07-08:
Hello,

Thank you for your help,

I changed my screen size to 1920x1080. The simulator still opens and directly closes.
This is my new Player.log

Mono path[0] = '/home/majd/Downloads/FlightGogglesv3-release/FlightGogglesv3_Data/Managed'
Mono config path = '/home/majd/Downloads/FlightGogglesv3-release/FlightGogglesv3_Data/MonoBleedingEdge/etc'
Preloaded 'lib_burst_generated.so'
Preloaded 'libassimp.so'
Display 0 'LEN T24i-10 24"': 1920x1080 (primary device).
Loading player data from /home/majd/Downloads/FlightGogglesv3-release/FlightGogglesv3_Data/data.unity3d
Unable to load player prefs
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=13
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.3.0f5 (44796c9d3c2c)
[Subsystems] Discovering subsystems at path /home/majd/Downloads/FlightGogglesv3-release/FlightGogglesv3_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=13
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: count=89
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_imageless_framebuffer, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance3, enabled=0
[... 52 lines cut ...]
[Vulkan init] extensions: name=VK_NV_dedicated_allocation_image_aliasing, enabled=0
[Vulkan init] extensions: name=VK_NV_device_diagnostic_checkpoints, enabled=0
[Vulkan init] extensions: name=VK_NV_fill_rectangle, enabled=0
[Vulkan init] extensions: name=VK_NV_fragment_coverage_to_color, enabled=0
[Vulkan init] extensions: name=VK_NV_framebuffer_mixed_samples, enabled=0
[Vulkan init] extensions: name=VK_NV_coverage_reduction_mode, enabled=0
[Vulkan init] extensions: name=VK_NV_geometry_shader_passthrough, enabled=0
[Vulkan init] extensions: name=VK_NV_sample_mask_override_coverage, enabled=0
[Vulkan init] extensions: name=VK_NV_shader_sm_builtins, enabled=0
[Vulkan init] extensions: name=VK_NV_shader_subgroup_partitioned, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_array2, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_swizzle, enabled=0
[Vulkan init] extensions: name=VK_NVX_device_generated_commands, enabled=0
[Vulkan init] extensions: name=VK_NVX_multiview_per_view_attributes, enabled=0
[Vulkan init] extensions: name=VK_NV_ray_tracing, enabled=0
Vulkan API version 1.0.0 (1.1.119 supported by driver)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce GTX 1080] id=[1b80]
Vulkan device type 2
Vulkan driverversion=[0x6e1d8080] uint=[1847427200]
Vulkan PSO: cache data not found [path=/tmp/MIT-AERA/FlightGoggles/vulkan_pso_cache.bin]
X Error of failed request:  BadLength (poly request too large or internal Xlib length error)
  Major opcode of failed request:  18 (X_ChangeProperty)
  Serial number of failed request:  226
  Current serial number in output stream:  227

varunmurali1, 2021-07-08:
I am not able to recreate this. Can you give me more information about your setup? What version of ubuntu are you running? Is this headless? Are you able to run `vulkaninfo` successfully? Is it possible you are running out of memory? Can you try running the binary with `./FlightGogglesv3.x86_64 -screen-quality 0`? Have you been able to run any version of FlightGoggles successfully with this system?

MajdWardeh, 2021-07-09:
Hi,

My setup is:
1. CPU: Intel® Xeon(R) W-2123 CPU @ 3.60GHz × 8 
2. RAM: 16GB.
3. GPU: GeForce GTX 1080 8GB of VRAM.
I am using Ubuntu 16.04.
There is one thing I'm not sure if it's relevant which is the GUI of the system was changed (by the IT staff of my university) to support RDP. If you would like to know more info regarding the GUI, I can get it for you.

I was able to run `vulkaninfo` successfully and the output is in the attached file [vulkaninfo_output.txt](https://github.com/mit-aera/FlightGoggles/files/6792039/vulkaninfo_output.txt).

I tried `./FlightGogglesv3.x86_64 -screen-quality 0` but got the same result (opens and directly closes) with the same error as the previous one presented in the log file.
Yes, I used to run previous versions of FlightGoggles on the machine. However, today I tried the older version `v2.0.4-b1`, and got the same error.
```
X Error of failed request:  BadLength (poly request too large or internal Xlib length error)
  Major opcode of failed request:  18 (X_ChangeProperty)
  Serial number of failed request:  226
  Current serial number in output stream:  227
```

Thank you for all the support.

Majd

varunmurali1, 2021-07-12:
The change to your X server is likely to be causing the issue. Can you run any other vulkan based application such as `vulkan-smoketest`?

MajdWardeh, 2021-07-12:
Hi, 

I was able to successfully run `vulkan-smoketest`. One important thing I noticed is that the system became laggy and it I open any window (a terminal for example) the system will lag a bit (including the `vulkan-smoketest`). 
[image]