mit-aera/FlightGoggles #167: Depth images for the gates

MajdWardeh, 2021-06-23:
Hello,

I was wondering if there is a way to change how a depth camera captures the gates? The resulted depth images don't have information about what is behind the gates. please check the screenshot.
Thank you,

[image]

varunmurali1, 2021-07-07:
The images seem to indicate a bug with the gate model. Let me check it out and get back to you.

MajdWardeh, 2021-07-16:
Hi, 

Any updates regarding the depth of the gates?

Thanks,
Majd

varunmurali1, 2021-07-17:
I haven't been able to isolate the issue yet. I will add a new binary when I am able to fix this.

MajdWardeh, 2021-07-20:
Great. Thank you so much!