mit-aera/FlightGoggles #170: Broken images in documentation

Ponop, 2021-07-18:
Currently the sections under "Vehicle Dynamics and IMU Model" are broken.
See here: https://flightgoggles-documentation.scrollhelp.site/fg/Multicopter-Dynamics.370999309.html

Below is how it appears in the webpage.
[image]

eatal, 2021-07-19:
Thanks for bringing this to our attention. The problem should be solved now. Please feel free to comment here, if you see any other issues.

closed 2021-07-19 by eatal

esiaero, 2021-07-31:
Hello, 
It appears [this](https://flightgoggles-documentation.scrollhelp.site/fg/Inertial-Measurement-Model.371163159.html) page is still(?) broken. 

example:
[image]

eatal, 2021-08-02:
I'm not able to recreate the issue anymore. As far as I can see, the equations are now rendered correctly. The system we use to generate the docs website seems to have some dependability issues though, so maybe the page looks different for you.

If you still cannot see the equations in the "Vehicle Dynamics and IMU Model" section, you can also have a look at the same description (with equations) in Section VI of the FlightGoggles article ([arxiv page](https://arxiv.org/abs/1905.11377)).

esiaero, 2021-08-02:
As of 8/1 It seems to have resolved itself, so it must have been a temporary issue. Sorry for the trouble