mit-aera/FlightGoggles #174: Stereo rendering in the provided Unity environment

MajdWardeh, 2021-09-05:
Hi,

Two weeks ago, I downloaded the Unity environment assets and followed all the instructions provided, and I built a custom environment (v2.0.5-smoketrail-1) that worked perfectly for a single RGB camera.
Now, I need a stereo camera, and it looks like the provided Unity environment has bugs regarding stereo-rendering or the use of a depth camera compared to the pre-built releases <= V2.0.4-b1.
Could you please give me some hints to solve this issue? I have zero knowledge of Unity, but I read your code and had some high-level understanding of it. 
I urgently need a stereo camera or an RGBD camera for my experiments, so your help is really appreciated.

Thank you,

varunmurali1, 2021-09-05:
Can you give me a little more detail about what you are trying to accomplish. 

Are you using the ROS client or the python client? What branch of the FlightGogglesClientBindings are you using? 

The current master branch is only intended to be used with v3.0 and might have backward compatibility issues. I suggest rolling back to `f25e1c2555d0c83f3cbac441d4e6973f09e2d0b9` for ROS with v2.x.

MajdWardeh, 2021-09-05:
I am trying to build an environment that has only squared gates in it. As I mentioned, I downloaded the assets and the  FlightGoggles renderer source code. I was able to build and run the environment.
Yes, I used an older version of ROS Client to get it running. I've read the source code for the ROS Client, and I know in depth what it does.
My problem is that when I ask for stereo-rendering, my built app will open with doubled height, as expected, but showing only one image that is stretched, not the two stereo images as expected. After the application has loaded the image, it crashes as soon as the drone moves.

varunmurali1, 2021-09-06:
Do you have a Player.log that I can look at?

MajdWardeh, 2021-09-06:
Hi, 

Thank you so much for your help.

I've just created a new project (on Windows), added the assets and the renderer code, built it (for a Linux system), and tested it on Ubuntu 16.04.

I built a new environment so that the line numbers in the Player.log file are consistent with the code that you have. since for my custom environment, I added a simple function to place the gates at poeses read from a txt file. 

Here's the content of Player.log for the new environment:
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=13
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2018.3.14f1 (d0e9f15437b1)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=13
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: count=116
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_acceleration_structure, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_buffer_device_address, enabled=0
[Vulkan init] extensions: name=VK_KHR_copy_commands2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_deferred_host_operations, enabled=0
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_imageless_framebuffer, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance3, enabled=0
[Vulkan init] extensions: name=VK_KHR_multiview, enabled=0
[Vulkan init] extensions: name=VK_KHR_pipeline_executable_properties, enabled=0
[... 1647 lines cut ...]
 #8 std::pow(float, float)
 #9 operator new[](unsigned long, std::nothrow_t const&)
 #10 start_thread
 #11 clone
 
(Filename:  Line: 307)

Receiving unhandled NULL exception
#0  0x00000000972517 in std::pow(float, float)
#1  0x000000009a8963 in std::pow(float, float)
#2  0x00000000ca30af in operator new[](unsigned long, std::nothrow_t const&)
#3  0x00000000ca31f1 in operator new[](unsigned long, std::nothrow_t const&)
#4  0x00000000c7a1e3 in operator new[](unsigned long, std::nothrow_t const&)
#5  0x00000000c7a3ef in operator new[](unsigned long, std::nothrow_t const&)
#6  0x00000000ac9e5e in operator new[](unsigned long, std::nothrow_t const&)
#7  0x00000000be6441 in operator new[](unsigned long, std::nothrow_t const&)
#8  0x00000000be70bf in operator new[](unsigned long, std::nothrow_t const&)
#9  0x000000009786af in std::pow(float, float)
#10 0x00000000c01b81 in operator new[](unsigned long, std::nothrow_t const&)
#11 0x00000000c02208 in operator new[](unsigned long, std::nothrow_t const&)
#12 0x00000000c022b8 in operator new[](unsigned long, std::nothrow_t const&)
#13 0x00000000c0243b in operator new[](unsigned long, std::nothrow_t const&)
#14 0x00000000cb22f0 in operator new[](unsigned long, std::nothrow_t const&)
#15 0x007fd31bef86ba in start_thread
#16 0x007fd31bc2e51d in clone