mit-aera/FlightGoggles #175: Could not find ./FlightGoggles.x86_64

ivsr, 2022-02-09:
I have finished installing FlightGoggles on Local Machine; however,   I could not find the binary file  _FlightGoggles.x86_64._ to run the Renderer.

Can anyone help me?

lukekenworthy, 2022-08-27:
I had the same issue. I believe you downloaded the source code from the [releases page](https://github.com/mit-aera/FlightGoggles/releases). Instead, you have to install from one of the Download Links on that page.

rlsch, 2022-10-26:
On the [release page](https://github.com/mit-aera/FlightGoggles/releases) above the **Assets** (Source code), there are 

"Download links:
[Ubuntu](https://drive.google.com/file/d/1NT2hkvJPDD1giI32k_zyISvp1bVBOSj9/view?usp=sharing)
[Windows](https://drive.google.com/file/d/1p25a5tzhxBoewxsJ2y_SHD2F5E_hS8QJ/view?usp=sharing)"

That's were the binary is.